hartk1213/MISC #4: Multiple potential fire-hazards

Kirinkunhime, 2021-09-14:
Good day!

After seeing a post on facebook about some issues about this board, I decided to give it a look and help out the community. As for my qualifications, I've designed the [Carabiner Toolboard](https://github.com/Kirinkunhime/Annex_Engineering_PCBs/tree/master/carabiner-toolboard) which is similar to the Doomboard in some areas.

But, that's not relevant, let me please point out a few issues with the current design that are literal potential fire-hazards that in the best case 'just' destroy the Doomboard, and in the worst case the mainboard of the printer.
They'll be listed from least severe to most severe.

First - The lower-left screw-hole. While not major, but the top 24V fill when combined with a washer could potentially cause the screw to carry 24V - this could short to the thermistorGnd. I'd recommend switching this over to a hole with a dedicated pad. While not fixing the issue with the washer it makes it more clear to people that there is no washer intended here.

Second - The HeaterGND trace. This is currently seemingly specced with 0.8mm. This means at an T_rise of 10° (10° over ambient temperature in the chamber) it could handle 2A. For a heater this is fairly low and considering that the Doomboard is inside a Voron (which are usually enclosed) and the extra enclosure will only trap heat inside even more - this will over time likely deteriorate the trace and will burn it out - leaving the board pretty much dead. As for this, I'd recommend increasing the width of the trace to at least 1.4mm (3A current capability with 10° T_rise).

Third - The screw-hole besides the 14-pin. This is what brought me here in the first place - due to not having a pad, the screw-head can very-likely scratch the coating of the HeaterGnd and ChamberThermistor. This could cause 24V to go to the thermistor and potentially cause major havoc there - depending on the board it might fry the MCU. As such I'd highly recommend switching this over to a padded hole - even though this will likely need massive trace re-routing.

Overall I'd recommend you to have another look at the traces - if e.g. the BAT-diode would be flipped 180° the routing from the top-left of the 14-pin would be much nicer and would make for a much cleaner route. 

I don't wanna say this will burn your house down - but if everything goes wrong, it could burn down someones house. As such I  strongly urge you to please fix those issues in a Rev 4.0 board as many Voron users seem to be using it, and having such potentially desastrous issues on a board with a sizeable userbase is not good for the whole 3dp community. 

If you wanna chat about those things, please contact me in Discord, iKirin#0001. I'll gladly help you to improve this board so we all can enjoy a better 3D-Printing space.

Best regards,
iKirin
PCB Hacker at Annex Engineering

hartk1213, 2021-09-14:
Alot of those issues have been addressed already...I don't know if the people buying/ selling the PCBs have used the updated Gerber files to not but I'll make sure to let them know there is a newer version to use... thanks for the feedback 👍🏻

closed 2021-09-14 by hartk1213

Coffee0297, 2021-09-14:
> 
> 
> Alot of those issues have been addressed already...I don't know if the people buying/ selling the PCBs have used the updated Gerber files to not but I'll make sure to let them know there is a newer version to use... thanks for the feedback 👍🏻

Looking at the newest gerbers here (V3.2) in JLC gerber viewer it seems like none of the above is fixed? or is there a newer version floating that is not here?

-Tonny

Kirinkunhime, 2021-09-14:
I am very sorry, but those issues are definitely not fixed, unless the Gerber for Version 3.2 is not up to date.

Let me attach screenshots for all of the issues, taken of the Gerber v3.2 for every single one of them.

First up - this is the screenshot with the version-number, rev 3.2
[image]

So, as for the first issue: Shorting potential with a washer. An M3 washer has a nomanal radius of 4.5mm - which has potential of shorting the trace here as mentioned:
[image]

For the second issue: That is straight up 0.8mm for the trace. 
[image]

And the third issue: An m3 socket head bolt has a diameter of 5.3 - 5.5mm, I've just drawn a rough 4.2mm diameter there but as you can see it'll easily touch the trace.
[image]

Unless there is a revision that is not yet uploaded to github, none of these issues are fixed, and this board still has the potential to harm people or damage property. As for now I've heard reports of a fried board - but if a lower quality board is used this could cause an Anet A8. And I strongly believe that no one in the 3d printer community wants more desasters on the scale of an Anet A8 to happen.

Best regards,
iKirin

engpedroamorim, 2021-09-14:
nylon washers can solve the problem?

hartk1213, 2021-09-14:
yes the latest version is not on github yet , im going over all of it again to triple check its all good to go before i push the new update

@Kirinkunhime 
@Coffee0297 

thank you !

matthewlloyd, 2021-09-14:
> I am very sorry, but those issues are definitely not fixed, unless the Gerber for Version 3.2 is not up to date / the latest revision.

Do you see issues in the ERCF version of the board? https://github.com/hartk1213/MISC/tree/main/PCBs/Afterburner_Toolhead_PCB/Production%20Files/ERCF

hartk1213, 2021-09-14:
@matthewlloyd  the ERCF board should be good to go as well, the issues were fixed on that one a while ago, i just never updated the standard one

Kirinkunhime, 2021-09-14:
Hi @matthewlloyd !

I've only had a quick glance over it so I can't promise stuff, but the ERCF version looks good to me - at least the 3 points listed in this github-issue are all fixed there.
There are some things I personally don't like with how some of the traces are routed, but that's a personal taste - and I'm very particular about those so ;)

hartk1213, 2021-09-14:
@Kirinkunhime @matthewlloyd 
all these changes were previously fixed like i thought, the changes were just in the voron hardware fork i had and i forgot to update this repo. both repos are in sync now and they are bothupdated to the latest 4.0