hartk1213/MISC #15: Z endstop sleeves

rlaferla, 2022-01-23:
The sleeves are nearly impossible to install.  I think you intended them to be heat inserted but the instructions make no mention of it.

rlaferla, 2022-01-23:
There is also no mention of what the connector is on the PCB!

NAPCAL, 2022-01-23:
The connector is a JST-XHP type, just like all the ones used on most all 3D-printer control boards. On the sleeves, press-fitting them is the best way to install; I wouldn't recommend heat setting them since this could cause the plastic to build up and block the movement of the pin or block the microswitch.

rlaferla, 2022-01-23:
I realized it was a 2-pin JST after creating this ticket.  I had a 3 pin (w/one unused) before.  Still, it should be noted in the docs.  Why are there two sleeves and not one?  It seems more error prone.  Again, it was impossible to press fit.  I used an ASA print.  Perhaps, there are implicit printing requirements - which also need to be documented.

NAPCAL, 2022-01-23:
I would guess that one sleeve would have been difficult to source. 

I wish the contact top was brass, so it would be resistant to plastics sticking to it.

rlaferla, 2022-01-23:
What tool do you recommend to insert them?

rlaferla, 2022-01-23:
Also, it looks like 1/4" (or thereabouts) copper tubing used in plumbing.  It's easily cut.

rlaferla, 2022-01-23:
Another question.  You should document w/photos how to insert these sleeves.  It may be easier if you insert one from the top and one from the bottom.  Maybe that's what you intended.  Inserting both from the top or bottom is hard.

NAPCAL, 2022-01-23:
Not a task most would take on. I got mine as a kit from KB-3D.com

rlaferla, 2022-01-23:
I did as well.  Sorry, I thought you were the author/creator.

NAPCAL, 2022-01-23:
I am not the creator of this mod just already have printed and assembled one.

rlaferla, 2022-01-23:
The seam between the two sleeves is problematic.

NAPCAL, 2022-01-23:
Wood clamps or small drill press work great for inserting the sleeves.

rlaferla, 2022-01-23:
Seems like he could have used 7mm OD 5mm ID PTFE tube instead of the copper.

hartk1213, 2022-02-21:
> The sleeves are nearly impossible to install. I think you intended them to be heat inserted but the instructions make no mention of it. I tried filing it to make it smoother but now the top sleeve is too loose. * purchased kit

no they just press in , i have no issues pressing them in on mine...i would check your extrusion , you might be over extruding a bit

closed 2023-01-17 by hartk1213