hartk1213/MISC #17: Issues with the Voron 2.4 GE5C Z joint BOM

mathiasm74, 2022-01-27:
The BOM for the GE5C Z joint calls for 4x M5x20, but these are two short. Without the "M5x1mm spacer" in place it just about reach the T-nut, but with the washer it's just too short. M5x25 screws do the trick.

The BOM also doesn't mention the 4 extra M3x16 screws that replace the original ones for the belt clip, as those are too long.

[image]

[image]

jakep82, 2022-01-27:
A more recent version of this is available on the VoronUsers repository which does call out the M3x12 screws necessary for the belt clip, and the M5x20 screws work fine for me using Igus bearings with the STLs at the link.

<https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/hartk1213/Voron2.4_GE5C>

mathiasm74, 2022-01-30:
The problem must be I didn't reprint the thinner belt clamps. With those I'm sure it works. 

Guess this is the best that could be done to adapt existing parts, but this design is flawed really. That single M5 screw really can't both clamp the belt clip and allow for any real movement in the joint. Hope the team will come up with a better solution for the next revision.

closed 2022-01-30 by mathiasm74