hartk1213/MISC #22: Voron2.4_Trident_Pins_Mod

shalev-d, 2022-04-02:
It seems that the CAD files are still 2.4r1, not as stated: "Hartk mod update for 2.4r2 (https://github.com/VoronDesign/VoronUsers/pull/638)"
Is there a forecast to when the 2.4r2 file uploaded?

bassco, 2022-09-18:
Furthermore, is there an updated seller listing or the seller's name for the outdated link to the rod supplier at [AliExpress](https://www.aliexpress.com/item/1739093502.html)?

hartk1213, 2022-09-21:
> Furthermore, is there an updated seller listing or the seller's name for the outdated link to the rod supplier at [AliExpress](https://www.aliexpress.com/item/1739093502.html)?

LinkCNC is the seller, there are also a bunch of vendors selling these as well

closed 2023-01-17 by hartk1213