hartk1213/MISC #23: PCBWay overlapped holes

jvandervyver, 2022-04-09:
Hi

I used your link to create an order of https://github.com/hartk1213/MISC/tree/main/PCBs/Stealthburner_Toolhead_PCB
I've got a lot of electronics experience but I've never ordered PCBs (usually I just etch my own).

I got a response from PCBWay as follows:

> Good day. Thanks for you PCB order#xxxxxxxxxxxx_Gerber_Stealthburner.

> There is an engineering question in your Gerber File. As shown in the attached screenshot,  there are 3 holes are overlopped and sizes are different. Which layer should we use? (drl or dril+1)        

> Please check and advise.

> The lead time will be re-calculated after the EQ is solved.

[image]

hartk1213, 2022-04-14:
I can update the Gerbers to fix that

hartk1213, 2022-04-14:
gerbers have been updated to fix that problem, ill close this issue now thank you for pointing that out

closed 2022-04-14 by hartk1213