hartk1213/MISC #27: Neopixel stops when Fan starts

SchwabJ, 2022-05-19:
Hello, 

is there something strange with GND? My PCF shut down the Neopixels at startup.

jamesc73, 2022-08-22:
I am having the same issue but only when the PCF is not 100%. Not sure if it's GND but I've tried changing that to a different terminal on my PSU and it made no difference. Just changed from an HARTK AB PCB and never had an issues with it.

HARTK SB 2 PART PCB

staticanime, 2022-11-12:
I'm having the same issue with the 2-piece PCB, has anyone else found a fix for this? Picture of the front portion is attached below.
LEDS come on when the system starts, as soon as I power up the part cooling fan the LED's cut out, can only turn them back on when PCF is off. Using the stock stealthburner_leds config included in my printer.cfg

For reference, I'm using:
LINNEO Switchwire Harness: https://www.onetwo3d.co.uk/product/voron-switchwire-heluflon-wiring-harness/?attribute_pa_color=black&attribute_toolhead-pcb=Yes
LINNEO Stealthburner LED Harness: https://www.onetwo3d.co.uk/product/linneo-stealthburner-internal-led-harness/?attribute_pa_color=natural-white-rgb&attribute_connector=JST+PH
GDSTIME GDB5015 for the part cooling fan: https://www.onetwo3d.co.uk/product/gdstime-5015-24vdc-blower-dual-ball-bearing/
GDStime GDA4010 for the hotend cooling fan: https://www.onetwo3d.co.uk/product/gdstime-gda4010-axial-fan-40x40x10mm-24v-2/

[image]

josephchh, 2022-12-31:
I have the same issue after installed the vorontap, it works perfect when I install the vorontap.

dawm, 2023-01-02:
Same issue here, linneo brand board, also have tap installed.
First/fresh build LDO kit, with all the wiring changes listed in the docs here.

hartk1213, 2023-01-02:
I personally have never had this issue, so I can't say for sure what is causing it...seems interesting for sure...I'll see if I can get it to reproduce and see what happens

islander2013, 2023-01-02:
had a similar issue also with the same 2 part pcb. however, mine would change color status for maybe 10 seconds and then randomly turn off....just the led.... then both the nozzle. however, it seems that by adding a in-line resistor near the mcu has made it more stable. not sure if this is a fix but saw it on the discord for the rainbow LED lights and thought would try.

dawm, 2023-01-02:
I've done everything I can think of, the ONLY thing that works is if i run a gnd wire from MCU all the way to the front PCB.
Was thinking maybe I had a bad ground wire in my harness but I tried a wire from the LDO breakout 14pin gnd pin to the toolhead 14pin gnd pin, no change. Also tried a jumper from toolhead 14pin gnd pin to front pcb (so its not my gnd wire) as well as the 100 Ohm at MCU. Still only works if I run a wire from MCU to front PCB. Possbily bad toolhead board? But I have continunity everywhere it needs to be.

josephchh, 2023-01-06:
I point the tap pcb 5-24V input +ve pin to SBPcba 5V output pin, the issue gone!

[image]

nibty, 2023-01-09:
> I point the tap pcb 5-24V input +ve pin to SBPcba 5V output pin, the neopixel works normal now.
> 
Thanks, this worked for me as well.

Following Hartk's suggestion and using the FS connector w/ 5v also solves the issue for me.
[image]
https://discord.com/channels/460117602945990666/1041007230239125574/1046975414880444498

arhodiewithsoul, 2024-07-14:
> I've done everything I can think of, the ONLY thing that works is if i run a gnd wire from MCU all the way to the front PCB bypassing the toolhead board. Was thinking maybe I had a bad ground wire in my harness but I tried a wire from the LDO breakout 14pin gnd pin to the toolhead 14pin gnd pin, no change. Also tried a jumper from toolhead 14pin gnd pin to front pcb (so its not my gnd wire) as well as the 100 Ohm at MCU. Still only works if I run a wire from MCU to front PCB. Possbily bad toolhead board? But I have continunity everywhere it needs to be.
> 
> * had an idea.. so I unplugged tap and it functions as it should.. guessing a new 24v tap issue :|

Please could I ask:

What MCU / board are using?
When you say you run a GND wire from the MCU - are you coming off the DCIN (-ve) for GND or thermistor (-ve) pin (AGND)? 

I'm experiencing similar LED issues on multiple MCU boards using 2pc Toolhead PCB's either with FAN0 and/or HE0. PWM signal causes SB LEDs to go off and only responds after PWM pins are no longer active. Z-probes are PL-08N/PL-05N on 24V.

Many thanks!

closed 2026-02-19 by SchwabJ