hartk1213/MISC #29: Wrong extruder pinout on  Schematic_StealthburnerPCB.pdf

JFD-Mtl, 2022-06-02:
The extruder pin order in Schematic_StealthburnerPCB.pdf  does not reflect the silk screen and does not reflect actual traces either.

The Extruder pin order should be
S1A/S2A/S1B/S2B
while it currently reads: S1B/S1A/S2A/S2B

This is not the standard order for 4 pin stepper connectors. 

Ideally, the traces should be updated to use the standard S1B/S1A/S2A/S2B that is normally used on 4 pin stepper connectors. From the gerber files, it does not look like there is anything preventing the correct pin order to be implemented.

hartk1213, 2023-01-17:
this has now been updated with the latest version pushed today 
revision 4

closed 2023-01-17 by hartk1213

AmigaZoid, 2023-02-11:
My PCB has S2B/S1B/S2A/S1A.  This must be wrong also right?