hartk1213/MISC #33: 2-Piece PCB Silk Screen mixup?

Durahl, 2022-06-28:
Today I upgraded from an AfterBurner 4.0 PCB to the New 2-Piece PCB StealthBurner and upon powering on the System both Tool Head FANs spun up to ( I assume ) MAX thus needing some investigation what went wrong despite me carefully paying attention if anything in the 14-Pin Cable Connector had changed ( besides the CT to 5V change ).

After some probing I found out that the problem being tied to how I soldered the FAN wires to the Secondary PCB based on Silkscreen labeling:
Given the options of **_24V, S, A1/A2, 5V_** & **_G_** - I established a connection via the **_24V_** and **_G_** Pads but the latter does **_not_** connect to the respective PCF and HEF traces coming from the 14-Pin Connector on the Primary PCB. Instead I had to use the **_S_** Pad ( for both Fans ) which I find rather... Unusual to say the least.

**Is this perhaps an Error in the Silk Screen?🤨**

[The store where I got my set from](https://lab4450.com/wp-content/uploads/2022/05/hartk-sb-2pcs-front-min.png) **_now_** populates the board with Dupont Headers which would have made it obvious but mine shipped bare ( and without the connectors ). Maybe I'm not the only one who got confused? 🤔

islander2013, 2022-07-05:
same here, i had no idea either...thank you for bringing this up.

bdistin, 2022-08-19:
PWM is switched on the ground side. I am not sure if S stands for signal or switched, but S would be right. Hopefully you didn't wire the duponts the same as in that picture. With the female on the breakout board there is a high risk of accidental shorting between the male pins on the main board. Ideally the breakout board should have the male pins as the breakout won't be energized until it seats into the female Dupont 9n the main board.

hartk1213, 2023-01-17:
> Today I upgraded from an AfterBurner 4.0 PCB to the New 2-Piece PCB StealthBurner and upon powering on the System both Tool Head FANs spun up to ( I assume ) MAX thus needing some investigation what [...]

if you were to look at the back of the PCB it tells you where each wire is supposed to go for 2-4 wire fans, 

this is not a error, S is the fans signal wire as its not just gnd. 
im going to close this issue as it doesnt seem to be an issue and just some confusion

closed 2023-01-17 by hartk1213