hartk1213/MISC #40: PINOUT_14_2 and Octopus wiring conflict

FlatEarthDave, 2022-09-19:
it doesn't seem like anybody actually solves these issues, but there is a serious conflict on the pinout_14_2 or whatever is named and the Octopus wiring diagram. the two are competently flipped upside down from each other in the 14-pin connector. which is correct?

hartk1213, 2023-01-17:
do you have a screenshot showing how its wrong?

Zephyrus67, 2023-03-06:
@hartk1213 Just open the Images\Wiring section of the resources you have made then you can see it...and I agree with Flat..it is rather confusing. It is about the 14_2_pinout.png and the SB_PCB_Octopus_wiring.png images. They have the 14 pin connectors reversed somehow..or one of the is turned upside down compared to the other.

Zephyrus67, 2023-03-06:
Never the less confusing :)

Zephyrus67, 2023-03-06:
since more than one had this experience it is not a bad idea to keep the thread here though. No one is arguing that you are wrong...but as I said..it can confuse...the thread with your new explanations will be helpful to others also getting confused.