hartk1213/MISC #63: V0.2 G2Z BOM unclear

RagingRoosevelt, 2025-01-15:
For the toothed idler, it's unclear which toothed idler is needed.  Is it 6/9mm 20T?  Or 6/9mm 16T?  Would either work?

acozad1, 2025-04-23:
Did you ever find this out? I was wandering the samething.

RagingRoosevelt, 2025-04-23:
I'm pretty sure it's supposed to be 2x of the 16T idlers:
- I installed a 20T and it feels like the belt is too close to the housing and will probably rub
- It looks like 20T is larger diameter than the 9mm called out in the BOM

Counterpoints:
- BOM says 1 of 9mm toothed idler beyond what is included in the G2Z BOM, and I'm pretty sure the one I got with G2Z is 20T
- CAD seems to show a 20T (not sure why I didn't think to check that earlier) 

[image]

RagingRoosevelt, 2025-05-10:
@acozad1, looks like 20T works.  It's closer than I'd like but it seems like it clears.